phatina/simple-mtpfs #37: HAVE_LIBUSB1 + libmtp-1.1.9: LIBMTP PANIC: Trying to dump the error stack of a NULL device!

jbeich, 2015-04-27:
libmtp [no longer](http://sourceforge.net/p/libmtp/code/ci/468e06b16c9aa87d90cb0ba7e75698038255540b) allow calling usb_init() more than once. At least on FreeBSD and libusb autodetected this results in
```
$ simple-mtpfs /mnt
LIBMTP PANIC: Trying to dump the error stack of a NULL device!
```
from
```
  * thread #1: tid = 100151, 0x0000000800cf4804 libmtp.so.9`init_usb + 4 at libusb1-glue.c:144, stop reason = breakpoint 1.1
      frame #0: 0x0000000800cf4804 libmtp.so.9`init_usb + 4 at libusb1-glue.c:144
      frame #1: 0x0000000800cef24c libmtp.so.9`get_mtp_usb_device_list(mtp_device_list=0x00007fffffffe0c8) + 28 at libusb1-glue.c:513
      frame #2: 0x0000000800ceeccf libmtp.so.9`LIBMTP_Detect_Raw_Devices(devices=0x00007fffffffe150, numdevs=0x00007fffffffe158) + 47 at libusb1-glue.c:633
      frame #3: 0x0000000000417d98 simple-mtpfs`MTPDevice::connect(this=0x0000000802419260, dev_no=0) + 88 at simple-mtpfs-mtp-device.cpp:94
      frame #4: 0x000000000040b5a4 simple-mtpfs`SMTPFileSystem::exec(this=0x0000000802419080) + 708 at simple-mtpfs-fuse.cpp:399
      frame #5: 0x0000000000417837 simple-mtpfs`main(argc=4, argv=0x00007fffffffe398) + 471 at simple-mtpfs-main.cpp:46
      frame #6: 0x00000000004047cf simple-mtpfs`_start(ap=<unavailable>, cleanup=<unavailable>) + 367 at crt1.c:78

  * thread #1: tid = 100151, 0x0000000800cf4804 libmtp.so.9`init_usb + 4 at libusb1-glue.c:144, stop reason = breakpoint 1.1
    * frame #0: 0x0000000800cf4804 libmtp.so.9`init_usb + 4 at libusb1-glue.c:144
      frame #1: 0x0000000800cf2c24 libmtp.so.9`configure_usb_device(device=0x00000008024213d0, params=0x0000000802419300, usbinfo=0x000000080242e090) + 52 at libusb1-glue.c:2008
      frame #2: 0x0000000800cc5347 libmtp.so.9`LIBMTP_Open_Raw_Device_Uncached(rawdevice=0x00000008024213d0) + 791 at libmtp.c:1887
      frame #3: 0x0000000000417efe simple-mtpfs`MTPDevice::connect(this=0x0000000802419260, dev_no=0) + 446 at simple-mtpfs-mtp-device.cpp:138
      frame #4: 0x000000000040b5a4 simple-mtpfs`SMTPFileSystem::exec(this=0x0000000802419080) + 708 at simple-mtpfs-fuse.cpp:399
      frame #5: 0x0000000000417837 simple-mtpfs`main(argc=4, argv=0x00007fffffffe398) + 471 at simple-mtpfs-main.cpp:46
      frame #6: 0x00000000004047cf simple-mtpfs`_start(ap=<unavailable>, cleanup=<unavailable>) + 367 at crt1.c:78
```

My workaround follows
```diff
# Reset libusb_context for successive init_usb() calls instead of error.
# Tested to not regress test_raw_devices.c from [patches:#64].

diff --git libmtp-1.1.9/src/libusb1-glue.c libmtp-1.1.9/src/libusb1-glue.c
index 2a6041f..7949115 100644
--- libmtp-1.1.9/src/libusb1-glue.c
+++ libmtp-1.1.9/src/libusb1-glue.c
@@ -142,7 +142,7 @@ static LIBMTP_error_number_t init_usb()
    * We use the same level debug between MTP and USB.
    */
   if (libusb1_initialized)
-     return LIBMTP_ERROR_NONE;
+    libusb_exit(NULL);
 
   if (libusb_init(NULL) < 0) {
     LIBMTP_ERROR("Libusb1 init failed\n");
```

phatina, 2015-04-27:
Thanks; I will investigate this soon.

karasuma-chitose, 2018-05-15:
The workaround doesn't work for me.