phatina/simple-mtpfs #38: libmtp not found on Mac although it is properly installed

GabrielGil, 2015-06-11:
First at all, I'd like to clarify you have to rename INSTALL to install.sh for example, so configure can start.

Then, configure is not able to find libmt although it is linked in /lib and can see my device using mtp-detect and even access the filetree..

I'd be up to help improving the quality on Mac OS X. I am using Yosemite.

Thanks

phatina, 2015-06-11:
Sure, feel free to contribute. I do not own any Apple product, so I'm not able to discover these issues.

pranav-prakash, 2015-07-05:
+1. I installed libmtp using brew but it still says `libmtp` not found.

Any suggestions?

pranav-prakash, 2015-07-05:
Ok I got it working in a super hacky way.

Go into `configure` file and change `ac_cv_lib_mtp_LIBMTP_Init` to be `yes` in both parts of the if statement.

Then manually change each of the header and cpp files in `src` to use `"/usr/local/Cellar/libmtp/1.1.8/include/libmtp.h"` instead of `<libmtp.h>`

It seems that brew doesn't symlink properly causing it to fail to find the header file.

Note that the instructions for installation could be a bit clearer.

You need `autogen` and `automake` installed and you have to start off with `./autogen.sh`

vrdriver, 2015-07-28:
+1 for the mac stuff. I hate to ask this, but is there any chance of posting/hosting an install package? I'm currently not in a position to compile at this point.

pranav-prakash, 2015-07-28:
@vrdriver I have a reasonably-working-on-osx fork at my repo:

https://github.com/pranav-prakash/jmtpfs

The installation instructions are also a bit more detailed. You will require a C++ compiler though. If you are on osx should you not have llvm by default (or if you don't you can install xcode command line tools).

vrdriver, 2015-07-28:
@pranav-prakash, I'll take a look. thanks.

pranav-prakash, 2015-08-16:
@vrdriver If you're using this to transfer files to an android device, also look into adbfs