phatina/simple-mtpfs #43: Add Moto X Pure to the device list

shague, 2015-12-05:
Device 0 (VID=22b8 and PID=2e82) is UNKNOWN.

Moto X Pure

[ 5136.557475] usb 3-2: new high-speed USB device number 13 using xhci_hcd
[ 5136.724405] usb 3-2: New USB device found, idVendor=22b8, idProduct=2e82
[ 5136.724412] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5136.724416] usb 3-2: Product: XT1575
[ 5136.724419] usb 3-2: Manufacturer: motorola
[ 5136.724421] usb 3-2: SerialNumber: TA3940071B

delcypher, 2015-12-05:
@shague I think this is an issue with libmtp and not simple-mtpfs.

See "New Device?"
at http://libmtp.sourceforge.net/

shague, 2015-12-05:
@delcypher your right. Sorry about that. I totally read that output wrong.

closed 2015-12-05 by shague