phatina/simple-mtpfs #47: Galaxy A5 2016 USB reset when trying to mount with simple-mtpfs

vitalif, 2016-04-10:
Hi! I don't know if this a problem of simple-mtpfs or the phone itself...
But when I'm trying to mount my galaxy A5 2016 (SM-A510F) (stock firmware) with simple-mtpfs, it results in a usb reset O_o.
dmesg from the phone is in the attachment.
[mtp_reset.txt](https://github.com/phatina/simple-mtpfs/files/211772/mtp_reset.txt)

MTP is obviously not mounted after reset, simple-mtpfs says:
LIBMTP PANIC: Trying to dump the error stack of a NULL device!
(this is because the device file in /sys/bus/usb vanishes)

This does not occur when mounting the phone with, for example, jmtpfs or android-file-transfer. But jmtpfs is very buggy and android-file-transfer fails to mount this phone read-write because it says "your device does not have android EditObject extension, mounting read-only".

Maybe it is possible to fix simple-mtpfs?

phatina, 2016-04-11:
Yes, it resets the device due to certain timeouts I encountered some time ago. See [mtp-device.cpp](https://github.com/phatina/simple-mtpfs/blob/master/src/simple-mtpfs-mtp-device.cpp#L199)... If this is the problem, I'll happily remove it, if it works correctly nowadays...

vitalif, 2016-04-11:
Interesting.. after removing both lines with smtpfs_reset_device it works. But why does it work with other phones with resets and does not work with A510F? O_o

phatina, 2016-04-12:
Once, I have some spare time, I will have a look at this, for sure. Thanks for the heads up.

jbeich, 2016-04-12:
Can you try libmtp workaround from #37 ?

vitalif, 2016-04-12:
Yes that's what I've done first :) it doesn't help.

tenforward, 2016-08-26:
I use Sony Xperia X (F5121). I have the same issue. I removed  https://github.com/phatina/simple-mtpfs/blob/master/src/simple-mtpfs-mtp-device.cpp#L199, but can't mount. I also removed https://github.com/phatina/simple-mtpfs/blob/master/src/simple-mtpfs-mtp-device.cpp#L153, then work correctly.

phatina, 2016-08-26:
Thank you guys! With testing and fixes.. I'll push these asap.

citrin, 2016-11-24:
I use Sony Xperia Z2 and can mount MTP only when smtpfs_reset_device on https://github.com/phatina/simple-mtpfs/blob/master/src/simple-mtpfs-mtp-device.cpp#L153 is removed.
smtpfs_reset_device on line 199 can be kept in my case.

commit fead6a76 (2016-11-27): Drop device reset (fixes #47)
closed 2016-11-27 by phatina