phatina/simple-mtpfs #48: LIBMTP PANIC: Trying to dump the error stack of a NULL device!

rtyle, 2016-10-14:
I have been using simple-mtpfs to rsync my music to my phone (AT&T Samsung Galaxy 6 (SM-G920A)) from Fedora Linux 23.
It has been working great (thank you!) ... until ...
My phone was updated ...
    Android 6.0.1
    Kernel 3.10.61-8851614
    Build MMB29K.G920AUCS5DPI1
    ...
... and things broke.

simple-mtpfs will list (-l) the device but on mount complains

    LIBMTP PANIC: Trying to dump the error stack of a NULL device!

I upgraded to Fedora 24.
simple-mtpfs-0.2-6.fc24.x86_64
No change.

I got the latest ...
    git clone https://github.com/phatina/simple-mtpfs
    commit ea08fdb1299a2f53ac758f15c0377434ac0bbf89
... and have been trying it.
I have seen the same error but most of the time I see

    Could not retrieve device storage.
    For android phones make sure the screen is unlocked.

My screen is unlocked but it asks

    Allow access to device data
    An MTP connection will be established to access the data on the connected device.
    DENY
    ALLOW

This all happens very quickly.

As a shot in the dark, I saw ...
https://github.com/phatina/simple-mtpfs/issues/47
... and tried something you suggested there ...
https://github.com/phatina/simple-mtpfs/blob/master/src/simple-mtpfs-mtp-device.cpp#L199
... to no avail.

I would be glad to help you fix/diagnose this problem but I have no idea what to do.
Help!?

----

I have always had a problem with my desktop gvfs mount colliding with simple-mtpfs.
The gfvs mount does not work for me (rsync).
I don't recognize any change/problem with gvfs (which is libmtp based too).
The phone asks to ALLOW this as well.
It still doesn't work for me.
To resolve the collision I have always used ...
    gvfs-mount --unmount $mount
... to unmount it ($mount, e.g. mtp://[usb:003,019]/) before trying simple-mtpfs.
I have also tried booting without graphics (multi-user.target) to no avail.

rtyle, 2016-10-15:
This seems related (although libmtp seems to work OK for me as a gvfs mount).
https://sourceforge.net/p/libmtp/bugs/1593/

rtyle, 2016-10-24:
See also
https://bugzilla.redhat.com/show_bug.cgi?id=1387921

onodera-punpun, 2017-01-11:
Same problem here...

smariel, 2017-12-01:
Try enabling USB debuging on your phone.
It solved this problem on mine.

oxguy3, 2018-01-25:
I'm having this issue as well, and because my device is not a phone (it's a GoPro Hero3+ Black Edition), enabling USB debugging is not possible.

kierun, 2018-11-29:
Ditto for me on Fedora 29. Used to work fine on Fedora 28 so something has change between those.

lukelbd, 2019-09-09:
Same issue. Unplugging the USB and then plugging it back in fixes the problem, but would like to see this fixed.

unInstance, 2020-01-08:
Try sudo. Worked out for me.

anddam, 2020-04-03:
I experienced this longstanding issue as well while trying to use simple-mtpfs, I saw the error is in accessing the `/dev/bus/usb/…` device.

The issue is not with simple-mtpfs then and should possibly be closed (it's quite stale anyway).

 

Just a note to avoid using sudo:

I added an `android` group, added my user to the group, then added a udev rule to get rw access:

```
# /usr/lib/udev/rules.d/98-android-mtp.rules
# Assign android group to my phone usb device

SUBSYSTEMS=="usb", ATTRS{idVendor}=="xxxx", ATTRS{idProduct}=="yyyy", GROUP="android", MODE="0660"
```

Use proper idVendor and idProduct values from `lsusb -v` output.

I noticed a few udev rules files tagging the device as uaccess and udev-acl but my system is not using systemd and from what I could read around systemd is supposed to handle those tags and grant users access to the devices.

crunchy101, 2023-03-22:
@anddam is correct, this is a permission error on the usb device. I solved this on Void Linux by adding my user to the plugdev group.