phatina/simple-mtpfs #49: Making a release for easy use on macOS

umireon, 2016-11-22:
Could you make a new release with the latest point, @phatina ?
Sadly only the very old version (0.2.0) is available on Homebrew 😢 
Since they won't also accept unversioned things, nobody can update your great product.

Thanks

phatina, 2016-11-22:
OK, will do.

umireon, 2016-11-22:
Thanks you so nice!

phatina, 2016-11-27:
You should see a new release, by now: [0.2.1](github.com/phatina/simple-mtpfs/releases/tag/simple-mtpfs-0.2.1)

closed 2016-11-27 by phatina

jbeich, 2016-11-27:
Can you retag it as `0.3` instead? In [FreeBSD package](https://www.freshports.org/sysutils/fusefs-simple-mtpfs) we already use `0.2.24` based on `0.2-24-ga7ab64c` from `git describe --tags`. While it's possible to switch to the actual tag by appending `,1`, I think, the amount of changes between `0.2` and `0.2.1` is similar to `0.1` vs. `0.2.1`.

phatina, 2016-12-25:
Done.