phatina/simple-mtpfs #53: Comparison with go-mtpfs

dandv, 2017-03-11:
Thanks for publishing this project. For users new to mtpfs, can you explain how it's different from @hanwen's older but more popular [go-mtpfs](https://github.com/hanwen/go-mtpfs) ?

jbeich, 2017-03-12:
Here's an **incomplete** list:
- go-mtpfs can use Android extensions
- simple-mtpfs supports more platforms (e.g. hanwen/go-fuse#85 or Go on sparc64)
- simple-mtpfs can mount by device file
- both support quick file access
- both were published in 2012

phatina, 2017-03-22:
When I have some time, I want to finish my work and use Android extensions, as well. But that code was created to learn something about fuse, android, and improve downsides of go-mtpfs... Nothing special, just a toy...