phatina/simple-mtpfs #58: Hang on cd DCIM/Camera

EmbeddedLinuxGuy, 2017-06-23:
Hi, I'm able to mount my Android device and move files to my host. However when I cd to DCIM/Camera, my shell hangs and I see these messages from simple-mtpfs -d:

```
getattr /
   unique: 213, success, outsize: 120
unique: 214, opcode: LOOKUP (1), nodeid: 1, insize: 45, pid: 19375
LOOKUP /DCIM
getattr /DCIM
   NODEID: 2
   unique: 214, success, outsize: 144
unique: 215, opcode: LOOKUP (1), nodeid: 2, insize: 47, pid: 19375
LOOKUP /DCIM/Camera
getattr /DCIM/Camera
   NODEID: 24
   unique: 215, success, outsize: 144
unique: 216, opcode: LOOKUP (1), nodeid: 24, insize: 45, pid: 19400
LOOKUP /DCIM/Camera/.git
getattr /DCIM/Camera/.git
```

Here are the messages I see when starting simple-mptfs -d:
```
zuz@ziz:~ (master)$ simple-mtpfs -d --device 1 ~/mnt
FUSE library version: 2.9.7
nullpath_ok: 0
nopath: 0
utime_omit_ok: 0
unique: 1, opcode: INIT (26), nodeid: 0, insize: 56, pid: 0
INIT: 7.26
flags=0x001ffffb
max_readahead=0x00020000
   INIT: 7.19
   flags=0x00000011
   max_readahead=0x00020000
   max_write=0x00020000
   max_background=0
   congestion_threshold=0
   unique: 1, success, outsize: 40
unique: 2, opcode: ACCESS (34), nodeid: 1, insize: 48, pid: 6301
   unique: 2, error: -38 (Function not implemented), outsize: 16
unique: 3, opcode: LOOKUP (1), nodeid: 1, insize: 47, pid: 6301
LOOKUP /.Trash
getattr /.Trash
   unique: 3, error: -2 (No such file or directory), outsize: 16
unique: 4, opcode: OPENDIR (27), nodeid: 1, insize: 48, pid: 6286
opendir flags: 0x18800 /
   opendir[0] flags: 0x18800 /
   unique: 4, success, outsize: 32
unique: 5, opcode: LOOKUP (1), nodeid: 1, insize: 45, pid: 19369
LOOKUP /BDMV
getattr /BDMV
   unique: 5, error: -2 (No such file or directory), outsize: 16
```
(etc)

EmbeddedLinuxGuy, 2017-06-26:
Could not reproduce after reboot.

closed 2017-06-26 by EmbeddedLinuxGuy

altblue, 2018-02-14:
Oh, I just stumbled upon your old issue report, and I have to say that, pitifully, this still happens all the time to me.
AFAICT rebooting and/or card re-plugging does not help.
It's always the same, **exactly** _3 minutes_ of seemingly nothing.
Here's a fresh example:
```
Feb 15 00:00:13 LOOKUP /Card/DCIM/Camera
Feb 15 00:00:13 getattr /Card/DCIM/Camera
Feb 15 00:00:13    NODEID: 6
Feb 15 00:00:13    unique: 43, success, outsize: 144
Feb 15 00:00:13 unique: 44, opcode: OPENDIR (27), nodeid: 6, insize: 48, pid: 27289
Feb 15 00:00:13 opendir flags: 0x18800 /Card/DCIM/Camera
Feb 15 00:03:13    opendir[0] flags: 0x18800 /Card/DCIM/Camera
Feb 15 00:03:13    unique: 44, success, outsize: 32
```
I'd like to understand why this happens, but I got no idea how to debug FUSE. :(

altblue, 2018-02-14:
Got rid of it using `go-mtpfs`: it (quickly) offered access to `/Card/DCIM/Camera`. There I found some files that `simple-mtpfs` didn't list. Moved those out of the way, umount, mount with `simple-mtpfs`, opendir takes no time. So it's definitely a `simple-mtpfs` issue, yet still no idea what. :(