phatina/simple-mtpfs #59: Phone and Card directories empty on Android 4.4.4

dcorking, 2017-08-17:
I mounted an Android phone but inside the mountpoint, I only find the Phone and Card directories, with nothing inside them.

```
$ simple-mtpfs phone
$ cd phone/
$ ls -laR
total 0
drwxrwxr-x  2 dcorking  staff  0  1 Jan  1970 Card
drwxrwxr-x  2 dcorking  staff  0  1 Jan  1970 Phone

./Card:

./Phone:

```

What have I done wrong?

The phone works with gvfs on Linux and with Android File Transfer on Mac OS X. (However on gvfs my filemanager mounts Phone and Card as separate mount points.)

```
$ simple-mtpfs --version
simple-mtpfs version 0.3.0
OSXFUSE 3.6.3
FUSE library version: 2.9.7
fuse: no mount point
$ simple-mtpfs --list-devices
1: SamsungGalaxy models (MTP)
```

dcorking, 2017-08-17:
Extract of output of `mount`

    simple-mtpfs@osxfuse0 on /Users/dcorking/phone (osxfuse, nodev, nosuid, synchronous, mounted by dcorking)

bsandrow, 2018-09-20:
Not sure what the real issue is here, but this can be fixed by using the `-f` flag to keep the process in the foreground. Not as convenient, but it _will_ get this working (at least it did for me).

dcorking, 2018-09-24:
@bsandrow that is a great workaround, thanks - foreground works perfectly for me on the same phone as when I posted the issue last year, and these versions:

```
simple-mtpfs --version
simple-mtpfs version 0.3.0
OSXFUSE 3.6.3
FUSE library version: 2.9.7
fuse: no mount point
```

The symptoms make me think the problem lies in the upstream fuse library.

dcorking, 2018-09-24:
I upgraded OSXFUSE and reinstalled simple-mtpfs from homebrew. The symptoms still occur without the foreground flag.
```
$ simple-mtpfs --version
simple-mtpfs version 0.3.0
OSXFUSE 3.8.2
FUSE library version: 2.9.7
fuse: no mount point
```
macOS version 10.13.6

githubjsorg, 2019-07-14:
> Not sure what the real issue is here, but this can be fixed by using the `-f` flag to keep the process in the foreground. Not as convenient, but it _will_ get this working (at least it did for me).

This workaround works for me also on my Galaxy S10.

Thank you to the developers who make it possible to manage files on my phone without using the terrible Android File Transfer interface.

lukelbd, 2019-09-09:
Any chance developers can address this? It's still a problem. I'd assume calling `simple-mtpfs` without `-f` is more or less the same as calling `simple-mtpfs -f &`, but the latter works and the former doesn't.

dcorking, 2019-10-26:
@lukelbd 
> The symptoms make me think the problem lies in the upstream fuse library.
https://github.com/phatina/simple-mtpfs/issues/59#issuecomment-423975149

dcorking, 2021-03-10:
I have now upgraded OSXFUSE by 3 minor points, and macOS to 10.15.7, with same symptoms. Anyone dug into this deep enough to be able to give upstream a reproducible bug report?
```
$ simple-mtpfs --version
simple-mtpfs version 0.4.0
OSXFUSE 3.11.2
FUSE library version: 2.9.7
fuse: no mount point
```

(Android File Transfer disconnects from my phone frequently with current versions.)