phatina/simple-mtpfs #60: ptp works but not mtp on fedora 26

slecrenski, 2017-09-16:
I have a Galaxy S8 phone that I'm simply trying to copy videos to my Fedora 26 machine.  I have the device connected over USB and ptp seems to work fine.  mtp does not.  It just hangs while trying to connect and when I run simple-mtpfs 

```bash
simple-mtpfs -l
1: SamsungGalaxy models (MTP)

simple-mtpfs /mnt
#no output and not mounted.

sudo simple-mtpfs /mnt
LIBMTP PANIC: Trying to dump the error stack of a NULL device!

uname -a
Linux MasterServer 4.12.9-300.fc26.x86_64 #1 SMP Fri Aug 25 13:09:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

cat /etc/fedora-release 
Fedora release 26 (Twenty Six)

sudo dnf install simple-mtpfs
Last metadata expiration check: 1:13:09 ago on Sat 16 Sep 2017 07:47:55 AM EDT.
Package simple-mtpfs-0.2-8.fc26.x86_64 is already installed, skipping.

dmesg | tail
[ 1533.851432] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851437] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851442] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851447] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851452] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851457] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851462] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851467] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851483] usb 5-2: usbfs: process 6115 (events) did not claim interface 0 before use
[ 1533.851604] usb 5-2: usbfs: process 10135 (simple-mtpfs) did not claim interface 0 before use
```
Even manually configuring everything did not work.
aka.  http://nknu.net/mount-nexus5-on-fedora23-using-simple-mtpfs/

slecrenski, 2017-09-22:
Or you could completely ignore this.  That works too.  

Last commit was about a year ago is this no longer maintained?

phatina, 2017-09-24:
Sorry, I am a bit busy at the moment. If you have patches, they are welcome... I want to find some time to give this project more love... :( Thanks a lot!

tandjelkovic, 2018-09-25:
Try calling simple-mtpfs with the -d switch and look in syslog (not dmesg). You have to install rsyslog first in newer Fedora versions. Here's what I get on Fedora 28 after connecting my Lenovo K6:

```
[root@thinkpad ~]# tail -f /var/log/messages
Sep 25 18:08:46 thinkpad kernel: usb 3-1: new high-speed USB device number 12 using xhci_hcd
Sep 25 18:08:46 thinkpad kernel: usb 3-1: New USB device found, idVendor=17ef, idProduct=7a2a, bcdDevice= 3.18
Sep 25 18:08:46 thinkpad kernel: usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 25 18:08:46 thinkpad kernel: usb 3-1: Product: Lenovo K6
Sep 25 18:08:46 thinkpad kernel: usb 3-1: Manufacturer: Lenovo
Sep 25 18:08:46 thinkpad kernel: usb 3-1: SerialNumber: e244976b
Sep 25 18:08:46 thinkpad mtp-probe[21524]: checking bus 3, device 12: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1"
Sep 25 18:08:46 thinkpad mtp-probe[21524]: bus: 3, device: 12 was an MTP device
Sep 25 18:08:46 thinkpad gvfsd[1267]: Device 0 (VID=17ef and PID=7a2a) is UNKNOWN in libmtp v1.1.14.
Sep 25 18:08:46 thinkpad gvfsd[1267]: Please report this VID/PID and the device model to the libmtp development team
Sep 25 18:08:46 thinkpad gvfsd[1267]: Error 1: Get Storage information failed.
```