phatina/simple-mtpfs #66: Error: Can not mount the device to '/Volumes/X'.

mrandrey, 2018-10-30:
I'm running simple-mtpfs on OS X with homebrew and while attempting to mount the device with "simple-mtpfs -v /Volumes/X", I get the following error: 
"Error: Can not mount the device to '/Volumes/X'."

I see the device in the list—"1: VariousViewpia DR/bq Kepler", /Volumes/X exists, running with sudo gives the same error. There's nothing in system log.

Any ideas?
Thanks.

mrandrey, 2018-10-30:
UPDATE: Got it to mount using the following command:
sudo simple-mtpfs -o allow_other,rw -v /Volumes/X

The disappearing folder on fail is a bit disconcerting, though.

cpcallen, 2019-05-31:
> If the folder is not in /Volumes, it seems to not be deleted afterwards.

That is the normal behaviour for unix `mount`: you have to create the directory (typically but not necessarily empty) before you can mount something on top of it, and when you unmount then the directory remains and whatever was in it (if anything) becomes accessible again.

It's macOS's `/Volumes/` which behaves weirdly.