phatina/simple-mtpfs #67: Unable to move a file

YuGiOhJCJ, 2019-06-07:
Hello,

I would like to move a file to a directory:
```
$ simple-mtpfs ~/mnt
$ cd ~/mnt/SanDisk\ SD\ card
$ mv miui_HM6AGlobal_V10.3.2.0.OCBMIXM_da9e59e3d0_8.1.zip zip
mv: cannot move 'miui_HM6AGlobal_V10.3.2.0.OCBMIXM_da9e59e3d0_8.1.zip' to 'zip/miui_HM6AGlobal_V10.3.2.0.OCBMIXM_da9e59e3d0_8.1.zip': Operation not permitted
```
Unfortunately, there is an error.

Hopefully, I can do that in two steps without using `mv`: 1) I copy the file to the directory 2) I delete the file.
```
$ cp miui_HM6AGlobal_V10.3.2.0.OCBMIXM_da9e59e3d0_8.1.zip zip
$ rm miui_HM6AGlobal_V10.3.2.0.OCBMIXM_da9e59e3d0_8.1.zip
```
Here it works.
I simulate `mv` by using `cp` then `rm`.

Of course, it is just a workaround and I am wondering why `mv` is not working.

Any idea please?

Thank you.
Best regards.

nathanshearer, 2020-11-27:
I can confirm this issue also exists on my Samsung Galaxy Note 9. Moving files does not work, but copy and delete does work.

jakubreron, 2021-06-06:
I also have this issue on my Mi 9T Pro, cannot move any file to the mounted device (I can copy files from the device without a problem)

DodoLeDev, 2023-12-31:
[You can probably find a solution here](https://github.com/phatina/simple-mtpfs/wiki#move-operation)