phatina/simple-mtpfs #79: possibly undefined macro: AC_DEFINE

NelsonSR, 2020-07-30:
I'm trying to compile the 0.4.0 release downloaded from GitHub, as I have done before with previous releases, but now it complains about an inexistent M4 directory and an undefined AC_DEFINE macro:

```  
debian:/usr/local/share/simple-mtpfs# ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:18: error: possibly undefined macro: AC_DEFINE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
debian:/usr/local/share/simple-mtpfs#
``` 
I'm runing Debian Buster 10.4

NelsonSR, 2020-07-30:
I've just tried to compile the previous version 0.3.0 getting the same error message:

```
debian:/usr/local/share/simple-mtpfs-simple-mtpfs-0.3.0# ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:18: error: possibly undefined macro: AC_DEFINE
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
debian:/usr/local/share/simple-mtpfs-simple-mtpfs-0.3.0#
```

borlafu, 2021-04-13:
You are probably using an old version of `pkg-config`. Updating it to `0.29.2` fixed this issue for me.

Ultra-Code, 2021-09-03:
As suggested by @borlafu  installing the latest pkg-config fixed this issue. For me I actually didn't have pkg-config installed at all so i just had to install it. But I think the error got when such a case happens isn't descriptive enough and can be improved