phatina/simple-mtpfs #80: Issue with editing files on mounted device

saandre15, 2020-09-21:
# What the Issues?
I am unable to edit files on any android device I've mounted. However, I'm able to create and delete files without any issue. Sometime when I initiate a file transfer the file is created on the mounted device, however the content is empty. Thus, I can deductively reason that the issue is limited to the inability for simple-mtpfs to edit created files.

 # How to replicate the issue? 
1. Mount the device
2. Go to the directory
3. ``` touch filename.txt```
4. ```echo file content > filename.txt```

## Expected behavior
No output
## Actual behavior
```echo: write error: input/output error```

# PC Config          
Build: https://aur.archlinux.org/packages/simple-mtpfs-git/
Distro: Arch Linux
Kernel: 5.8.9-arch2-1

# Android Device Config
Oneplus 8 128GB

dgulotta, 2020-10-13:
I am having the same issue.  I am using Gentoo Linux and a Moto G6.  The issue started with commit 2bcc4551e9ad7c3827a52af7edd1888775ab5acc; everything worked fine before that.

GOKOP, 2021-08-03:
I have the same problem. What's more interesting, the device is useless after that error happens. I can read from it fine before it, but after the first failed write any file operation at all hangs the program performing it. Furthermore, after unmounting the device it can't be mounted again (LIBMNT PANIC: trying to dump stack of a NULL device!) until I disconnect my phone and connect it again.