phatina/simple-mtpfs #81: Hangs or errors writing to Google Pixel from macOS

cpcallen, 2020-10-28:
I am attempting to copy files to a Google Pixel from macOS using `simple-mtpfs`.  Unfortunately, almost everything I have tried fails.  I can usually create the target file, but never write any bytes to it.  Often the writing process hangs; inevitably I am unable to terminate either it or `simple-mtpfs` itself using `^C`.  Occasionally it hangs badly enough to cause other unrelated filesystem activity to hang until I physically disconnect my phone from the USB cable.

```
 ~$ simple-mtpfs -f -o rw,volname=Pixel ~/Pixel &
[1] 11810
~$ df Pixel
Filesystem            1024-blocks    Used Available Capacity iused ifree %iused  Mounted on
simple-mtpfs@osxfuse0   117354836 7274368 110080468     7%       0     0  100%   /Users/cpcallen/Pixel
~$ touch Pixel/test
~$ ls -l Pixel/test 
-rw-r--r--  1 cpcallen  staff  0 28 Oct 15:19 Pixel/test
~$ echo "Hello" > Pixel/test
```
At this point echo hangs.   I press ^C many times.  After a long (~minute) delay:
```
-bash: echo: write error: Input/output error
^C^C^C^C^C^C^C^C^C
~$ umount Pixel/
~$
```
The volume is unmounted, but `simple-mtpfs has not terminated.
```
~$ fg
simple-mtpfs -f -o rw,volname=Pixel ~/Pixel
^C
~$ 
```

I can delete files just fine (provided I've not tried to write to any since I mounted the volume):
```
~$ simple-mtpfs -f -o rw,volname=Pixel ~/Pixel &
[1] 12408
~$ ls -l Pixel/test 
-rw-r--r--  1 cpcallen  staff  0 28 Oct 15:27 Pixel/test
~$ rm Pixel/test
~$ ls -l Pixel/test 
ls: Pixel/test: No such file or directory
~$ umount Pixel/
[1]+  Done                    simple-mtpfs -f -o rw,volname=Pixel ~/Pixel
~$ 
```
Note that `simple-mptfs exits of its own accord after the volume is unmounted as it should.

Sometimes, instead of just hanging, the filesystem will report "no space left on device" even though (for example) I'm only trying to copy a <100KiB file onto a device with >100GiB free:
```
~$ simple-mtpfs -f -o rw,volname=Pixel ~/Pixel &
[1] 5035
~$ df Pixel
Filesystem            1024-blocks    Used Available Capacity iused ifree %iused  Mounted on
simple-mtpfs@osxfuse0   117354836 7323692 110031144     7%       0     0  100%   /Users/cpcallen/Pixel
~$ cp Music/Clips/Android/Ringtones/Western\ Electric\ 554.mp3 Pixel/Ringtones/
# After a long delay:
cp: Pixel/Ringtones/Western Electric 554.mp3: Input/output error
cp: Music/Clips/Android/Ringtones/Western Electric 554.mp3: could not copy extended attributes to Pixel/Ringtones/Western Electric 554.mp3: No space left on device
```

I have had no difficulties writing files to the device using [Android File Transfer for Linux](https://github.com/whoozle/android-file-transfer-linux).

I'm running this on macOS 10.13.6 (17G14033), connecting to a Google Pixel (original) running Android 10 (build QP1A.191005.007.A3).  I installed `simple-mtpfs` and its dependencies using Homebrew; output from `simple-mtpfs --version`:
```
simple-mtpfs version 0.4.0
OSXFUSE 3.9.0
FUSE library version: 2.9.7
fuse: no mount point
```

og900aero, 2020-11-26:
Same problem on pure Arch Linux with Oneplus 7 and simple-ftpfs. Go-mtpfs works fine.

sir-marv, 2021-01-10:
Same issue on arch with simple-ftps & motorola g7 pro. Works fine with my old nexus 5.

JakeSFR, 2022-11-04:
This commit seems to be the culprit: https://github.com/phatina/simple-mtpfs/commit/2bcc4551e9ad7c3827a52af7edd1888775ab5acc

Problem confirmed on my HP Touchpad (CyanogenMod/KitKat-4.4), somebody else's unspecified phone with Android 4.4 and Oppo Reno Z (Android 11).

All of them report:

```
# mtp-detect | grep -i partial
101b: Get partial object
95c1: Get Partial Object (64bit Offset)
95c2: Send Partial Object
```

what makes the new method introduced in the above commit to kick in, resulting in hangs and errors.

My old Samsung phone with Android 4.2.2 has only the "Get partial object" cap, so simple-mtpfs falls back to the old method and it works fine there.
Reverting the commit makes it work also on HP Touchpad.

Fatdog64-813/libmtp-1.1.17/simple-mtpfs-0.4.0.

msalle, 2023-01-08:
I can confirm that reverting that commit https://github.com/phatina/simple-mtpfs/commit/2bcc4551e9ad7c3827a52af7edd1888775ab5acc makes two phones (MotoG7 with Android 10 and Fairphone FP3 with Android 11) work again with an otherwise unpatched 0.4.0. Both report the 3 partial object capabilities.
0.3.0 works also fine for both of them.

encore64, 2024-02-11:
Can also confirm.

I have a Samsung Galaxy S9 with Android 10 where I've used simple-mtpfs together with rsync with success for several years.

I tried to do the same today with a Samsung Galaxy S24 with Android 14 and rsync seemed to get stuck as soon as it wrote files larger than 0 bytes. Reverting the commit 2bcc455 on an otherwise latest version of simple-mtpfs fixed the issue.

Letus252, 2025-01-30:
I had the same issue. The problem was with kde dolphin automounting mtp devices. I had no idea how to disable it so I went to "/usr/lib/qt6/plugins/kf6/kiod" and changed kmtpd.so permissions to read only (using chmod 000 kmtpd.so).  (Credit to OP of this [post](https://www.reddit.com/r/kde/comments/mvrtck/restarting_kiod/) for workaround).