phatina/simple-mtpfs #83: Error while installing

zzzMarrezzz, 2021-04-14:
Hi!
I get the following errors while installing on Ubuntu 20.04
`sudo ./autogen.sh
utoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
autoreconf: Leaving directory `.'

sudo ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
./configure: line 3477: syntax error near unexpected token `noext'
./configure: line 3477: `AX_CXX_COMPILE_STDCXX_17(noext)'
`
Please advise.

EpicOrange, 2021-05-14:
Exact same error here, on macOS 10.13.6 (High Sierra)

EpicOrange, 2021-05-14:
Ah, this worked for me!

1. install the latest `autoreconf-archive` package
2. in the repo toplevel, run `autoreconf -i`
3. `cd build/` and try `../configure` again

IOACHIMVS, 2021-06-04:
On GNU/Linux, for instance being in simple-mtpfs' toplevel:

1. Download GNU Autoconf Archive
`$ git clone git://github.com/autoconf-archive/autoconf-archive`
2. Move GNU Autoconf Archive's m4 directory to simple-mtpfs' toplevel
`$ mv autoconf-archive/m4 .`
3. Run `autoreconf -i`
4. Run `cd build/` and `../configure`