phatina/simple-mtpfs #84: LIBMTP PANIC: Trying to dump the error stack of a NULL device!

Cyp, 2021-08-21:
- Trying to mount a phone (Samsung Galaxy S21 Ultra) with simple-mtpfs-0.4.0:
```
$ simple-mtpfs -dfs mount_directory/
LIBMTP PANIC: Trying to dump the error stack of a NULL device!
```
- It has worked previously with the same device before a system upgrade. Might have been using simple-mtpfs-0.3.0 before.
- Trying with a different utility (which doesn't work either):
```
$ go-mtpfs mount_directory/
2021/08/21 12:57:30 fatal error LIBUSB_ERROR_TIMEOUT; closing connection.
2021/08/21 12:57:30 OpenSession failed: LIBUSB_ERROR_IO; attempting reset
2021/08/21 12:57:31 selectStorages failed: EOF
```
- This pops up a message on the phone:
```
Allow access to phone data?
The connected device will be able to access data on this phone.
```
- If not clicking allow, and trying simple-mtpfs again, no error is printed, but it still doesn't work.
- If clicking allow, then simple-mtpfs prints the `LIBMTP PANIC: Trying to dump the error stack of a NULL device!` error again.

rh-jelabarre, 2021-09-23:
Having the same issue here, Fedora 34 (Cinnamon Desktop), to a Motorola Moto e5 Play 

Name        : simple-mtpfs
Version     : 0.2
Release     : 17.fc34

Tried against USB2 and USB3 ports

Gragt, 2022-03-29:
Have you tried unmounting the phone in Nautilus and then mount it with `simple-mtpfs`? This worked on Fedora 35.

mcepl, 2023-07-10:
It is some access control problem, but it would be nice if `simple-mtpfs` at least said what file it wanted to access and what group was expected the user to be in.

derekatkins, 2024-06-14:
For what it's worth I was just having this issue.  I unplugged my phone from the computer, plugged it in again, turned on MTP, and it worked.  Not sure where the issue was, but that solved /this/ issue.

jip, 2025-02-20:
The command
```console
sudo killall kiod5
```
launched before mounting solved the problem for me. Found [here](/JasonFerrara/jmtpfs/issues/19#issuecomment-685705205) .