phatina/simple-mtpfs #90: there is no configure file

AstroShipt, 2023-05-09:
when i tried to do the ../configure it says "no such file" hm

phatina, 2023-05-09:
Run `autoreconf -fi` first. It'll generate `configure` script for you.

chefexperte, 2023-07-17:
I also got the same error, and `autoreconf -fi` did not work for me. However running `./autogen.sh` fixed the problem. 
Also, maybe it's a Fedora-only problem, but libmtp is not what I needed to build, I also needed libmtp-devel.

SCBuergel, 2024-01-09:
> I also got the same error, and `autoreconf -fi` did not work for me. However running `./autogen.sh` fixed the problem. Also, maybe it's a Fedora-only problem, but libmtp is not what I needed to build, I also needed libmtp-devel.

thanks for this - I would not have been able to manage without this. I suggest to update the README. Also I needed to install `autoconf` as a dependency.

Unfortunately, the next step fails: https://github.com/phatina/simple-mtpfs/issues/92

derekatkins, 2024-06-14:
@antofthy install autoconf-archive for those extra macros.  Then configure runs cleanly (at least on Fedora 38)

antofthy, 2024-06-17:
My summary of MTP commands I have found or looked at, including simple-mtpfs (no longer available on fedora 40).
My specific focus is on Unison file transfer (sync), and vim text file editing, rather than photos and music.

https://antofthy.gitlab.io/info/misc/phone_file_transfers.txt

derekatkins, 2024-06-17:
@antofthy --
> My summary of MTP commands I have found or looked at, including simple-mtpfs (no longer available on fedora 40). My specific focus is on Unison file transfer (sync), and vim text file editing, rather than photos and music.
> 
> https://antofthy.gitlab.io/info/misc/phone_file_transfers.txt

You missed adbfs in your list.

I've been using simple-mtpfs and rdiff-backup to backup my phone to my linux laptop for the past several years, but the MTP-based solutions were timing out this past week (I had some 2+GB files to transfer).  After simple-mtpfs failed, I tried gvfs (which DOES mount into a directory, by the way), and I believe I tried one other solution before arriving at adbfs -- which worked!!!  And worked quickly..

For reading from phone it worked -- i didn't try anything else.

Still, it is sad state at how bad this all is.

antofthy, 2024-06-18:
@derekatkins
> You missed adbfs in your list.

Thanks I took a look.
   https://github.com/zach-klippenstein/adbfs
  
 Looks very experimental, usage not explained very well, especially for
 someone not familiar with the 'Android Debug Bridge', or Go programming.
 
> I tried gvfs (which DOES mount into a directory, by the way),

You are right it does mount, but like many other MTPfs mounts very limited.
You can't rename files for example!

I found  `android-file-transfer`  which is in the fedora repos to work better
that simple-mtpfs (which is no loner in the repos), as it does some
weird things, which DOES allow you to rename a file OVER an existing file.

> Still, it is sad state at how bad this all is.

Totally agree.   Wish newer phones allow normal USB filesystem mounts again,
rather than limiting it to this MTP crap!

derekatkins, 2024-06-18:
@antofthy Actually I used https://github.com/spion/adbfs-rootless.git

> Totally agree. Wish newer phones allow normal USB filesystem mounts again, rather than limiting it to this MTP crap!

Agree.  My FIRST android phone attached as a USB Disk.  It was fantastic.  Then they moved to MTP and I was like -- WTAF?  Why add a layer here?  Although from a security PoV it makes sense.

antofthy, 2024-06-19:
Actually it wasn't security POV... the phone can handle that.

It was that USB mounts can leave a file systems in a bad way if you unplug at a bad moment.  MTP fixes that, but was designed for media files (music and photos) not whole directory file transfer, and file renaming, which is where MTP filesystems fail.