phatina/simple-mtpfs #92: `../configure` fails

SCBuergel, 2024-01-09:
I downloaded your latest release (0.4.0) and tried to do the pre-buiild step which fails, I'm on a clean Debian:

```
user@work:~/simple-mtpfs-0.4.0$ ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force 
aclocal: warning: couldn't open directory 'm4': No such file or directory
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
autoreconf: Leaving directory `.'
user@work:~/simple-mtpfs-0.4.0$ cd build/
user@work:~/simple-mtpfs-0.4.0/build$ ../configure 
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for g++... g++
checking whether the C++ compiler works... yes
checking for C++ compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of g++... gcc3
../configure: line 3505: syntax error near unexpected token `noext'
../configure: line 3505: `AX_CXX_COMPILE_STDCXX_17(noext)'
```

SCBuergel, 2024-01-09:
this can be omitted by including `sudo apt install autoconf-archive` as a dependency. The README should be updated to include this.

requeijaum, 2024-12-12:
Debian 'testing' here.

```bash
checking dependency style of g++... gcc3
../configure: line 3858: syntax error near unexpected token `noext'
../configure: line 3858: `AX_CXX_COMPILE_STDCXX_17(noext)'
```

Still the same problem in a different line.